Ultimaker/Cura #17324: Printer profile 'Anycubic Kobra Max' overrides speed stting.

nhc155, 2023-11-15:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Anycubic Kobra Max 

### Reproduction steps

I did test prints with a simple rectangular print, at speed 20/40/60/80/100 mm/s and found almost NO difference in print times (all between 16 and 17 minutes), The gcode produced in all cases seems to force a speed of 100mm/s.

I created a new profile from scratch. With this profile, speed 20 took 50:01 minutes, and the 100 too 15:36 minutes.

I also have Creality Ender 3 V2 and there is no such issue with the profiles for it.
I have attached gcodes for 20 and 100, using the default values from the profile - cura sees them both taking 26 minutes with a .4mm nozzle size.
I have also attached an export of the profile.

[AKMAX 20.txt](https://github.com/Ultimaker/Cura/files/13362570/AKMAX.20.txt)
[AKMAX 100.txt](https://github.com/Ultimaker/Cura/files/13362571/AKMAX.100.txt)
[Anycubic Kobra Max #2.csv](https://github.com/Ultimaker/Cura/files/13362572/Anycubic.Kobra.Max.2.csv)

### Actual results

All prints took the same time regardless of speed setting.

### Expected results

No change.

### Add your .zip and screenshots here ⬇️

[AKMAX 20.txt](https://github.com/Ultimaker/Cura/files/13362606/AKMAX.20.txt)
[AKMAX 100.txt](https://github.com/Ultimaker/Cura/files/13362607/AKMAX.100.txt)
[Anycubic Kobra Max #2.csv](https://github.com/Ultimaker/Cura/files/13362608/Anycubic.Kobra.Max.2.csv)

GregValiant, 2023-11-15:
Thanks for the report.
Could you try the same tests but with "Speed / Flow Equalization Ratio" set to 0.0?

I'm also getting odd speed results in 5.5 when the setting is at 100%.  With all speeds set to 100mm/sec I'm getting speeds in the 190 range in the gcode.  It appears to be worse than it was in 5.4.
G1 F7768.2 X149.421 Y58.661 E1102.43771
G1 **F11390.8** X151.597 Y58.66 E1102.47534
G1 **F11501.4** X151.597 Y58.454 E1102.47887
G1 **F11390.8** X149.421 Y58.453 E1102.51651
G1 F7768.2 X148.887 Y58.344 E1102.53033
G1 F5923.2 X148.659 Y58.095 E1102.54156

nhc155, 2023-11-15:
I set the "Speed / Flow Equalization Ratio" to visible, and it was set at 100%, so I ran a slice at 20mm/s.I changed it to 0% and ran another slice at 20mm/s. Both estimated the print at 27 minutes, so no effect !
I compared both gcode files using Notepad++ and they were identical except for the final 2 lines, which are  comments:100%:    ;SETTING_3  pla\\nsetting_version = 22\\nposition = 0\\n\\n[values]\\nspeed_prin    ;SETTING_3 t = 20.0\\n\\n"]}
0%:    ;SETTING_3  pla\\nsetting_version = 22\\nposition = 0\\n\\n[values]\\nspeed_equa    ;SETTING_3 lize_flow_width_factor = 0\\nspeed_print = 20.0\\n\\n"]}
Cheers,Neil.

GregValiant, 2023-11-15:
In previous versions I noticed that Equalization was having an effect on the speeds but nothing like what I'm seeing now.  I generally print with the "Infill Layer Height" at 2X layer height.  That makes the infill go down every other layer, but at twice the flow rate.  Cura would compensate by _lowering_ the speed to keep the mm³/minute in line with the other features.  That cause the Infill to print at lower speeds.
I guess part of what is happening (in the case of my model) is that my inner wall line width is being adjusted and so Cura is speeding up the nozzle to compensate for the lower flow.
I did a print speed test once and my printer can go to about 175 but those 190's I see in the gcode are beyond what it is capable of.  In addition, if I have the Max Feedrate X and Max Feedrate Y set to 100 then there should not be any speeds in the gcode in excess of 100.

nhc155, 2023-11-15:
I don't know what's going on with the profile that Cura has for the Anycubic Kobra Max. Something in it is off. When I created one from scratch, no problems with the speed.
I reported it in case it was a problem with Cura 5.4 .0 but I'm sure the issue is with the profile, although maybe it is a problem with both.
Regards,Neil.
Yahoo Mail: Search, organise, conquer

nhc155, 2023-11-16:
Just installed Cura 5.5.0, same problem with that profile, but I'm happy with the one I created from scratch.
Cheers.

GregValiant, 2026-10-01:
I'll close this as being stale.  This problem was addressed at some point.

closed 2026-10-01 by GregValiant