Ultimaker/Cura #17326: bonnie mask

X-raul8, 2023-11-15:
### Cura Version

5.0.0

### Operating System

windows 11

### Printer

tenlog tl-d3-pro

### Name abnormal settings

si

### Describe model location

no

### Describe your model

no

### Add your .zip here ⬇️

[Bonnie Mask Complete.zip](https://github.com/Ultimaker/Cura/files/13365939/Bonnie.Mask.Complete.zip)

GregValiant, 2023-11-15:
Thanks for the report.
Without a project file I can't tell how you set up your slice.  The model does have several errors.  I wouldn't expect the teeth to print correctly without repairs to the model.
This is from Mesh Mixer.
[image]

This is from forware.co/onlinestlrepair:
-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 5427 Non-manifold edges (?)
--> 6780 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 4347 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 47312ms ------
-> Vertex count changed from 105889 to 84150 (-21739)
-> Triangle count changed from 217572 to 168736 (-48836)
That probably doesn't mean much to you, but it isn't good.

This is the repaired model in 5.5.0.  It's an 84 hour print and 860 grams of filament.
[image]

I'm going to remove the bug label from this.  With all the serious errors in the model there isn't any way to tell if it doesn't slice because of a bug, or because of the errors.  I sliced it in 5.5 because I don't have 5.0 loaded anymore.  You should consider upgrading to 5.2.2 as it is the most stable of the 5.x Cura versions.

closed 2023-11-15 by GregValiant

X-raul8, 2023-11-15:
fine, I ready upgrade to 5.5