Ultimaker/Cura #17328: The image in cure does not correspond to the one supplied as stl.

Matorra, 2023-11-15:
### Cura Version

5.4.0

### Operating System

windows 10

### Printer

Artllery Genius

### Reproduction steps

I try to laminate an stl file and it does not correspond to the original. I try to do it in Prusa_slicer and it does it correctly. This has happened to me several times with other files and sometimes it has been corrected but I don't know why.
[image]
[image]

### Actual results

I have NOT been able to get it to generate correctly.

### Expected results

[image]

### Add your .zip and screenshots here ⬇️

[AG_remates Vinco-servo.zip](https://github.com/Ultimaker/Cura/files/13367540/AG_remates.Vinco-servo.zip)

GregValiant, 2023-11-15:
Thanks for the report.
Go to the Mesh Fixes settings and turn off "Remove all Holes".

I'll go ahead and close this as it is a settings issue.
[image]

closed 2023-11-15 by GregValiant

Matorra, 2023-11-15:
Thank you very much, I will try it as soon as I can.Enviado desde mi Galaxy
-------- Mensaje original --------De: GregValiant ***@***.***> Fecha: 15/11/23  19:39  (GMT+01:00) Para: Ultimaker/Cura ***@***.***> Cc: Matorra ***@***.***>, Author ***@***.***> Asunto: Re: [Ultimaker/Cura] The image in cure does not correspond to the one supplied as stl. (Issue #17328) 
Thanks for the report.
Go to the Mesh Fixes settings and turn off "Remove all Holes".
I'll go ahead and close this as it is a settings issue.