Ultimaker/Cura #17329: Slice failed with unknown error

zeroturnzed, 2023-11-15:
### Cura Version

5.5

### Operating System

Windows 11

### Printer

Anycubic i3 Kobra Plus

### Name abnormal settings

no, no

### Describe model location

When I move it no. When I rotate it 90 degrees, yes, it slices then.

### Describe your model

It is a bracket to mount a decoration on the front of the hotend shroud.

### Add your .zip here ⬇️

[AKP_Kobra Faceplate Mount.zip](https://github.com/Ultimaker/Cura/files/13368102/AKP_Kobra.Faceplate.Mount.zip)
[Kobra Faceplate Mount.zip](https://github.com/Ultimaker/Cura/files/13368105/Kobra.Faceplate.Mount.zip)
The AKP Kobra Faceplate Mount is the successful slice and the Kobra Faceplate Mount is the original .stl file. As the .stl is oriented it will not slice. Once I lay the object flat then Cura will successfully slice it. Settings are supports = true, Build Plate Adhesion = raft, Speed = 90, Material =200/55, Quality =.2, Top/Bottom = 4, Walls 1.2mm/3. I wouldn't consider this situation a "crash". It was just an incomplete function. I was able to continue without rebooting the software, so it was just an error trapping function of the software that prevented a crash.

GregValiant, 2023-11-16:
Thanks for the report.
Try rotating the model and/or moving it around on the Cura build plate and/or changing the Initial Layer Height by a little bit.  There is a bug in 5.5.0 that can deep a model from slicing but one or more of those workarounds might fool the bug.
This sliced for me on the first try without any problem.
[image]

zeroturnzed, 2023-11-18:
Since putting 5.5.0 on my computer I have been getting a lot of layer shifting that destroys the print. It does it on every print since. I've followed all of the suggestions for curing layer shift with little success so far. I have calibrated, layer meshed the bed, PID, reduced print speed down to below 50mms, I've tightened all the bolts nuts and screws, made sure the belts are snug, lubricated the moving parts. I put a print on to print about an hour ago that I sliced with the previous version of CURA and using Pronterface to send it to the printer and the print is right on target so far with no shift at all. I have had to postpone all of the orders I have pending until I can guaranty that I can fulfill the orders soon. I will probably lose some of the business to other print shops. 

5.5.0 has issues, it looks good but it doesn't perform well and this is just not with one printer. All 5 of my printers stopped printing with any quality after upgrading CURA. I'm going to remove it from my computers and download an older version or just switch to another slicer all together.
Steven Imlay

GregValiant, 2023-11-18:
@zeroturnzed I have a piece of code that I wrote that will read a gcode file into AutoCad.  If you would post one of those "bad" gcode files I'll load it and check it.
So far, of the 200 or 300 files from other people that I've checked (for various reasons), and the 100 or so files of my own that I've checked, there has never been a single "layer shift" in the gcode.  I have never experienced a layer shift in my own printing with any version of Cura.  Although I prefer 4.13.1, I use the latest version of Cura because I hang around here triaging bug reports.

The slicing bug seems to have appeared in 5.4.0 with a vengeance.  The Cura team has worked on it but it is still a real problem in 5.5.0.
Nobody likes buggy software.  You don't, I don't, and I'm sure the Cura team doesn't.  The slicing bug will get chased down and eradicated.  Since the release of 5.4.0 and now into 5.5.0, that _"I don't want to slice this"_ bug accounts for about 95% of the bug reports here.  This is not the first report I've seen that is attributing "layer shifting" to Cura.  If it is true (post that gcode) then it would be the first time I will have seen it and I've been triaging here for a couple of years.

zeroturnzed, 2023-11-19:
Unless, I am keeping a gcode for future printing they get deleted good or bad after the end of the print job. So I don't have any of the bad prints to share. I have already deleted 5.5.0. I added 4.13.1 to see what it is like and deleted all other slicers except Prusaslicer. I only use Prusa when I need to test a slice that is questionable. I find it difficult to work with for everyday use. As I expected there were only 2 of my printers in the cura printer list so I had to create custom printer profiles for the rest. I need consistent results and good quality prints. I am trusting 4.13.1 to slice a print that has to be delivered tomorrow. So, we will see how it does with custom printer profiles.

Steven

GregValiant, 2026-10-01:
I'll close this as stale.

closed 2026-10-01 by GregValiant