Ultimaker/Cura #17330: Printing Problem after slicing... Plinter freezes

majopro77, 2023-11-15:
### Cura Version

5.5.0

### Operating System

win 10

### Printer

Anycubic Kobra 2 Plus via CFFFP

### Reproduction steps

I opened a File to slice it for my printer... everything gone perfekt... but while Printing after about 25% the Printer stoppet moving left the Nozzle inside a wall and started cooling the printer... all Motors where OFF... printer freezes..touchpanel out of control.. have to take the power off and on... 
 i tried it out.. it only happens with files lagerer than 150x150 mm... 

### Actual results

read the text "Reprodution steps"

### Expected results

read the text "Reprodution steps"

### Add your .zip and screenshots here ⬇️

[AK2P_Faltkorb gross.zip](https://github.com/Ultimaker/Cura/files/13368980/AK2P_Faltkorb.gross.zip)

Asterchades, 2023-11-16:
The entire GCode file is there and appears to be in good order, which leads me to suspect this may be a printer issue rather than a Cura one. Most likely the printer entered a "panic" state which in turn caused it to shut down as a failsafe measure.

As to what caused it to panic? Ordinarily this would be (or, at least, should be) accompanied by a message on the screen telling you (even if vaguely) what went wrong. But in the absence of such feedback it would be simply a guess as to what occurred. My instincts would be towards either thermal runaway protection (which should have brought up a message) or an under-voltage spike causing the controller to enter a partially-rebooted state.

MariMakes, 2023-11-16:
Hey @majopro77,

Welcome to the UltiMaker Cura Github 🚀 
I'm agreeing with @Asterchades here, it seems like an issue with your printer, and not with Cura. 

To me, it doesn't seem like a bug in UltiMaker Cura. 
Would you still like to keep this issue open or do you agree that we can close it?

majopro77, 2023-11-17:
HiI was contacting you because I fixed the problem for me… I’m now using the older cura 5.4.0.  there it works!!!! And that was why I contacted you!!!So I think it use something that come from the difference between cura 5.4.0 and 5.5.0If you want… I can add the stl I was useing..So you can   Slice it.Lg John+++++++++++++++++++++++++John RosenauVon meinem iPhone SE2 / MacBook pro gesendetSend via my iPhone SE2 / MacBook pro

jsfrerot, 2023-12-09:
Also had freezing issues. Anycubic support told me to use anycubic slicer and no freeze happened since then.

GregValiant, 2026-10-01:
I'll close this as stale.

closed 2026-10-01 by GregValiant