Ultimaker/Cura #17331: Failed slice when using tree supports

brekkeso, 2023-11-15:
### Cura Version

5.4.0

### Operating System

Windows 10 Enterprise

### Printer

Creality Ender-5

### Name abnormal settings

seems to be tree supports with several models. sometimes moving on bed helps. This time it didn't

### Describe model location

didn't help this time

### Describe your model

crossbar relocator and raspPi camera mount for extrusion

### Add your .zip here ⬇️

[CrossbarRelocator_DX-Failed slice.zip](https://github.com/Ultimaker/Cura/files/13369372/CrossbarRelocator_DX-Failed.slice.zip)

Vandresc, 2024-01-19:
👋 Quick update on our side.
Your issue seems to have been resolved with this fix from our wonderful community member ThomasRahm!
(https://github.com/Ultimaker/CuraEngine/pull/2002)

I'll close this issue since it's resolved, a fix should be available with the upcoming release.

Thanks again, and please let us know if you run into any other issues 💪

closed 2024-01-19 by Vandresc

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0