Ultimaker/Cura #17335: file failed to slice

jsilvius, 2023-11-16:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Creality CR-10 S5

### Describe model location

no

### Describe your model

no

### Add your .zip here ⬇️

[CCR1_R5D4 Dome Single Print MKIII (with details) v2.zip](https://github.com/Ultimaker/Cura/files/13372073/CCR1_R5D4.Dome.Single.Print.MKIII.with.details.v2.zip)

jsilvius, 2023-11-16:
> Thanks for the report.
> It sliced for me so I can't duplicate the problem. You can try rotating
> the model or making subtle changes to the Initial Layer Height.
>
> The Cura team will take a look.
>
> I think I'd enable "Make Overhangs Printable" for this. There are some
> flat roofed pockets on the outside and the outer edges are curved. That
> won't work even with the Bridge Settings enabled.
> [image: image]
> <https://user-images.githubusercontent.com/64202104/283610826-752ae6e9-2d76-4ad8-9123-9e12b0411d2a.png>
>
> This is the first layer over one of the pockets.
> [image: image]
> <https://user-images.githubusercontent.com/64202104/283611555-980e0f81-4624-4883-9924-c1583107f36e.png>

Thank you Greg.
I've tried it on three different computers now, so I'm not sure what I'm
doing incorrectly.
I've been using the Cura software since v3, and have never had this issue
before.
I've never seen the " Make Overhangs Printable" setting. When I turned it
on, it asked for a maximum setting, do you have a suggestion?

Thank you again,
James

------
James Silvius

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic