Ultimaker/Cura #17337: Slice failed

MakeItMaketMakeIt, 2023-11-16:
### Cura Version

5.5.

### Operating System

Windows 10

### Printer

My AMBP, a scratchbuilt i3

### Name abnormal settings

Can't use Default, not applicable.

### Describe model location

rotated 45 across the plate

### Describe your model

Yes. Yes. Fixed Simple Holes and fixed Normals as I have done before, Yes, Still fails, no changes

### Add your .zip here ⬇️

[V3_19X_Standard.zip](https://github.com/Ultimaker/Cura/files/13377182/V3_19X_Standard.zip)

MakeItMaketMakeIt, 2023-11-16:
I have successfully sliced this before but now it just fails. No changes, nothing diff afaik.

GregValiant, 2023-11-16:
I get an error message "The zip folder is empty".

MakeItMaketMakeIt, 2023-11-16:
Apologies, but s’ok now, I put it on the plate along the Y axis and now it slices and prints.

 

Thanks anyway.

 

Allen

MakeItMaketMakeIt, 2023-11-21:
Hello Greg,

 

I’m on Cura 5.5, Windows 10 64bit

I own 3 machines: a Monoprice Maker Select V2 that has been heavily modded, a scratchbuilt 200 x 200 i3, and another scratchbuilt i3 with a 300 x 300 bed, all on Marlin 1.1.9 with TMC2208 drivers.

 

When I last replied to you, it was in the affirmative that I was successful in slicing a particular part.

Well, it seems like that issue is continuing to rear its ugly head.

Couple things here:

The model is not watertight, so attempting to fix it with Mesh Tools (Fix Simple Holes, Fix Model Normals) takes a long time and sometimes Cura hangs up.

On the rare occasion that I am successful in Mesh tool application, it still fails the slice.

 

 

This is puzzling to me, as I have sliced and printed a couple dozen frames already and now it’s just being difficult.

This only is occurring when I print frames, either horizontally, vertically, or diagonally across the build plate. No other part that I have printed has caused me any of these issues, just frames.

I am happy to provide you with any info you wish.

I am away from my pc for the better part of the day, so we may be going back and forth as this gets looked at.

 

The 3mf is too big for an email.

You can get it from my secure cloud here: https://mega.nz/file/IzJCDKxL#NAT8e3tm31XBAiaAWo2QP4qf3mEfli5J9i69GOuPavE

 

Please advise accordingly.

 

Thank You, Greg.

 

Allen

GregValiant, 2023-11-21:
I downloaded your file and it looked OK but Cura is telling me there are "no models in the file" and MS 3D Builder throws an error when I try to open it there.

Moving on.
I call the bug the "Petulant Child" child bug.  It's like Cura is saying "No...I'm not going to slice this one".  It appears that certain models on certain systems just refuse to slice.  I haven't been able to get a better handle on it than that.  A couple of things I have noticed:

- Engineering models are a lot less likely to be affected than "organically" shaped models.
- Models that have lots and lots of vertices within a couple of layers (EX: buildings with rough hewn wood floors or rough tile floors) either take forever to slice, or fail.  That happens a lot.
- Some models are being produced at such high resolutions that the slicer struggles with the sheer density of the mesh.  This is a new thing and likely more prevalent in scanned models.

I did come across an interesting 3rd party article that explored the possibility that the problem is caused by computer manufacturers routinely over-clocking the processors in computers.  It was interesting, and I brought it up with the Cura team but right now there is no way to check that.  It might all be BS anyway.  I'm not throwing it out because a surprisingly large number of models that won't slice on someone else's system slice fine on my Win10Pro basic laptop system with just the laptop monitor.

Probably 95% of the bug reports here are in regards to the Petulant Child bug.  The other bug reports are often either setting or model problems (or people just venting).  I know the UltiMaker team is looking at this problem as it affects their printers as well.  It is slightly better from 5.4 (bad) to 5.5 but still seems to be present in 5.6beta2.

Cura always evolves.  Maybe an added feature broke something down the line.  If you need to slice those frames and "variable line width" isn't required I would suggest you install 4.13.1 and use that.  Of the 5.x versions I think 5.2.2 is the most stable.

MakeItMaketMakeIt, 2023-11-21:
Wow!

What a marvelous, informative answer. Really.

Thank You. 😊 (not being snarky here, I’m serious) 😊

Learn something new every day, sometimes twice in the same day. 😊

 

Ok. that’s twice now, I put something on the plate and save it as a 3mf and send it to you but it’s empty.

Ok, if Cura says it’s not there in the 3mf, here’s the link to the 26X STL on my same server. Zipped, s’too big for email.

https://mega.nz/file/BrRUxQRb#XVlxlGQ-JgKwp4OPDSGKS_lq_CfD9oBXMEKctZcnNhc

Let Cura try to tell you it can’t see it NOW. 😊

 

I’ve found that a large number of the frames I’ve printed so far required some mesh work with Curs, holes and normal was all it took.

I struggled with a 19X in the original email, but repositioned it along one axis and it worked.

Still trying with the 26X as I type this, fixing it in 5.2.2 as you suggested (was already on my PC) then will try a slice.

Ok, tried slicing along the x axis, failed about 30 seconds in.

File is in the link. Back to you. 😊

 

Variable Line Width.

Not familiar with it’s use, but I recall reading about that somewhere.

Is it “standard” now or is it configurable?

I searched Settings for variable and came up empty.

Might just be automatic now…

 

Thank You, Greg.

 

Allen

closed 2025-08-27 by GregValiant