Ultimaker/Cura #17344: Large items always crash *DUE TO* large number of tree supports.

tlhintoq, 2023-11-17:
### Cura Version

5.5.0

### Operating System

Win10x64pro

### Printer

CR10s-5 (500mm³)

### Name abnormal settings

Only happens with tree settings on large models

### Describe model location

Orientation is not a factor other than reducing the number of supports

### Describe your model

Model is confirmed good through 3 other tools as well as mesh fix

### Add your .zip here ⬇️

[BackLeft_settled.zip](https://github.com/Ultimaker/Cura/files/13386828/BackLeft_settled.zip)

Cosplay armor - Back of full body wearable armor.  So larger than your average trinket.  
These larger items common fail with tree supports.  Sometimes increasing the support angle in order to reduce the number of supports can get around it, or putting in blockers to reduce number of supports.  But it clearly is number of tree supports that's doing it.

tlhintoq, 2023-11-17:
Interesting follow up...

After the fail to slice... After saving successfully and lodging the report above...
Tried to close the job before opening a new one and got this:
[image]

Maybe related - maybe not.

5axes, 2023-11-17:
Edit : solution uncheck **Print Thin Walls**

tlhintoq, 2023-11-17:
> Should be based on an existing profile.
As you can see, it is.  I'm not even sure how I could make a profile that _isn't_ based on an existing profile since that's how Cura works and forces the creation of them.  I mean, when you save a profile you get a dialog that says "This is inheriting from blahblah" or however its worded.
[image]
Is it possible that you don't have the profiles in between a base that ships with Cura and the one I'm using? For example:

1. Default -> CustomPETG
2. CustomPETG -> CustomAvoidEcho
3. CustomAvoidEcho -> PETG armor

So if you don't have the entire lineage of profile inheritance would you then get that "not supported" comment that I don't get?

> Edit : solution uncheck Print Thin Walls
I am totally stunned that worked - but I'll confirm it did.  I'm guessing this is some kind of known bug for you to have tried it and found it in a matter of a couple hours.  Going from number of supports to `[  ] print thin lines` isn't exactly an intuitive jump because they seem related.  I know I wouldn't have gone from one to the other.
It does open a concern then.  If this is some kind of conflict where its trying to support a thin wall only because `[x] Print thin walls' is ticked on - does this mean it works because it no intention of printing that thin wall and thus the part will be missing a wall?  Its kinds feeling like its a choice: EIther have supports or have the thin wall but can't have both.  And that's going to mean  the job can't be run because its going to fail for either reason A or reason B.

MariMakes, 2024-01-18:
Hi @tlhintoq,

Quick update from our side 👋
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪

Even with the changes we made in the latest versions of Cura we still see that your model crashes. 🫤 But we have added your project file to our analyzing tool and found there were at least 10 crashes that were similar to yours, which helps us to narrow down what the cause is of your crash so we can resolve this for you and others.
Hopefully, we can find a solution soon 🤞

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/17438

closed 2024-01-18 by MariMakes

tlhintoq, 2024-01-18:
Ok.  FYI:  Gave up on Cura like so many others.

Y'all can't get the one part to slice without crashing.  Yet same computer with another slicer can do that part and 3 others equally big - all over 1m triangles EACH - for the new 800x800x1000 OrangeStorm.

I used to blame the Arachne engine because Cura went bad when that was introduced.  Yet other slicers are running the same engine fine.  But since generation 5 of Cura its just... well... being nice I'll just say "its history".
[image]