Ultimaker/Cura #17345: Slicing keeps failing, even after fresh install

Jan-Snijder, 2023-11-17:
### Cura Version

5.5.0

### Operating System

windows 11

### Printer

ender 3 pro

### Name abnormal settings

none

### Describe model location

standard

### Describe your model

benchy trireme

### Add your .zip here ⬇️

[CE3PRO_Trireme_nosupports.zip](https://github.com/Ultimaker/Cura/files/13390227/CE3PRO_Trireme_nosupports.zip)

After a full fresh install (as asked by marimake) I sliced this benchy which I have sliced many times before on older versions, but Cura still gives an error on slicing.

I hope someone finds this too long lasting bug and gets Cura up and running for ALL of us :(

As Cura V 5.3.1 does work and slices perfectly I am 1000% shure its a bug, and not a small one, that is causing this, because as of newer versions the problem was introduced and I for one have not changed anything in my config since.

GregValiant, 2023-11-17:
Thanks for the report.
My printer is also an Ender 3 Pro.  This is in 5.5.0:
[image]

This bug (and it is indeed a bug) seems to have shown up in 5.3.1 and was really bad in 5.4.0.  There are three possible workarounds that alone or in combination can often fool the bug.

- Move the model away from the Cura midpoint.
- Rotate the model.
- Change the Initial Layer Height by a little bit (like from 0.20 to 0.23).

I'll mark this as a duplicate.  I'll leave it open for the Cura team to take a look.

Jan-Snijder, 2023-11-18:
As I have already tries the solutions you mention with no effect I am dead in the water here and am bound to conclude Cura has left the race for me, I'll wait for a solution (that will hopefully come someday, when devs decide to go from scrap as of version before 5.3.1 and redo  their steps to find out where it gone wrong, but I doubt they will and keep on traveling a road thats ending in a dead end and the burrial of Cura but thats their choice)
In the meantime I have to stick with older versions to be able to use Cura on my Enders as long as I can.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

Jan-Snijder, 2024-04-03:
I tried to install again and slice some parts, but still not slicing anything here, as 5.6.0 and 5.3.1 do slice all parts I tried I Guess the bug is still there, otherwise it would slice as 5.6.0 does

HellAholic, 2024-04-03:
[image]
[image]
[image]
I just sliced your model with tree support and without tree support in 5.7.0 on two different machines. I don't think there is an issue in Cura here since Greg also managed to slice the it in 5.5

If you had the crash again in 5.7.0 we will receive the report via sentry if your device is connected to internet and you gave Cura the permissions to access network.

Jan-Snijder, 2024-04-03:
> image.png (view on web)
> image.png (view on web)
> image.png (view on web)

I did so the report should get sentSent from Android device