Ultimaker/Cura #17347: No time estimation available when opening a .gcode file

php4fan, 2023-11-17:
### Cura Version

5.4.0

### Operating System

Manjaro Linux

### Printer

irrelevant

### Reproduction steps

1. Go to "File->Open File" and open a .gcode file 
2. Look at the time estimation

### Actual results

There is no time estimation

### Expected results

When I generated this very same gcode file using Cura itself by slicing a STL model, there was a time estimation. So I should be able to get a time estimation for an already existing .gcode file. Actually I opened the gcode file for that very purpose, because when I created it I didn't write down the time estimation.

If estimating the time is a time-consuming operation (which can be done with less overhead when slicing, and that's the reason for having the estimate only when you slice), then  there should be an "estimate time" button near to where the "No time estimation available" text is. I have looked everywhere in the menus and UI and I can't find anything. If an option does exist to estimate the time for the gcode, it's too difficult to find.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-11-17:
Thanks for the report.
There is always a time estimation.  The only occasions I've seen that particular warning come up is when a slice completed but there were no layers, no material usage, and consequently no time lapsed.  That can be because of model errors, or model geometry (the entire model was too thin to slice).
Without a project file there can be no investigation.

php4fan, 2023-11-17:
> Without a project file there can be no investigation.

That's not true, that depend on your will to investigate.
But fortunately I can share a file that reproduces the issue for me
[CE3S1_ifs-new-batt.zip](https://github.com/Ultimaker/Cura/files/13395585/CE3S1_ifs-new-batt.zip)

Start a new project and load this file.
But this happens to me with ANY of the gcode files that I have, al generated with Cura itself.

GregValiant, 2023-11-17:
Now see, without that file I wouldn't be able to tell you that Cura does not "read" gcode files, just displays them.  That's why it can't show the time estimate because it doesn't _know_ the time estimate.  When you slice a file it's an entirely different thing.

If you want to see the time you need to open the file in a text editor and translate the line ";TIME:12284" which is usually the second line in the file and the units is "seconds".

I'll remove the bug label from this as it is expected behavior.  You could put in a feature request for this.

closed 2023-11-17 by GregValiant