Ultimaker/Cura #17351: Slicing failed when activating support for small part

hd42, 2023-11-17:
### Cura Version

5.5.0

### Operating System

Ubuntu Linux

### Printer

Anycubic i3 Mega

### Name abnormal settings

Support

### Describe model location

No

### Describe your model

It is watertight

### Add your .zip here ⬇️

[AI3MSPRO_HaubenHalter.zip](https://github.com/Ultimaker/Cura/files/13392711/AI3MSPRO_HaubenHalter.zip)

5axes, 2023-12-07:
Tree support Issue .. similar to https://github.com/Ultimaker/Cura/issues/17549

Part not appropriate for tree support 
[image]

Vandresc, 2024-01-19:
👋 Quick update on our side.
Your issue seems to have been resolved with this fix from our wonderful community member ThomasRahm!
(https://github.com/Ultimaker/CuraEngine/pull/2002)

I'll close this issue since it's resolved, a fix should be available with the upcoming release.

Thanks again, and please let us know if you run into any other issues 💪

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

hd42, 2024-04-03:
Works for me, thank you!