Ultimaker/Cura #17352: Some seam settings cause lines to start in mid air

mbenkmann, 2023-11-17:
### Cura Version

5.5.0

### Operating System

Linux

### Printer

Creality CR-10S4

### Reproduction steps

1. Slice the attached model with layer height 0.2 and seam preferences as seen in screenshot (note particularly Seam Corner Preference: None)
2. Look at layer 128. The first line (green) being printed starts (and ends, because it's a closed loop) over thin air
3. Change Seam Corner Preference to "Smart Hiding", "Hide or Expose" or "Hide Seam"
4. Look at layer 128. The first line correctly starts at a point that is supported by the layer underneath
5. Note: Seam Corner Preference: "Expose Seam" produces the same error as "None"

### Actual results

Depending on the Seam Corner preference the print partially fails.

### Expected results

No matter what the Seam settings are, Cura should never start/end a line in thin air when supported alternatives exist. I assume that there is some prioritization going on inside Cura where the Seam settings override the choices Cura would otherwise make regarding where to start/end the line. But this should never cause an usupported start point to be chosen over a supported one. This always causes inferior prints and is therefore never what the user wants. The typical use for seam settings is to get a cleaner looking print. It is counterproductive if the seam settings cause slicing results that by the laws of gravity will always be inferior. In this particular case it's especially annoying because the line in question is on the inside of the model, so it's not even a line that the user particularly cares about regarding the seam (because it won't be visible). This causes the situation where the user is forced to choose between a print that looks bad on the outside but is mechanically fine, or a print that looks nice on the outside but is mechanically bad inside.

### Add your .zip and screenshots here ⬇️

[PrintInAir.zip](https://github.com/Ultimaker/Cura/files/13394043/PrintInAir.zip)
Project file with the model and settings

[image]
Screenshot of incorrect behavior (green line starts in thin air). Note the Seam Settings.

[image]
Screenshot of correct behavior (green line starts supported by prior layer).

GregValiant, 2023-11-17:
Thanks for the report.
If you are going to try to print this without supports you need to enable the bridge settings.  That will change the behavior.  If there were supports under that area then this would be a non-issue.
In addition you have the Z-seam location in the middle of the back.  That is causing Cura to start that line in the middle of the back which for this model is over air.  If you change the Z-seam location to Back-Left or Back-Right the problem will go away.
This appears to be a settings issue.  See if any of those fixes this for you.

mbenkmann, 2023-11-17:
You did not read my report. I am well aware that it is the settings that cause Cura to begin the line in mid air. The point is that Cura SHOULD NOT start the line in mid air even with these settings.

GregValiant, 2023-11-17:
I read your report.  The settings you are using don't work together as you have found out.  That model would typically be printed with support.  You have chosen not to use support but you have also chosen not to use the Bridge Settings which would fix it, or to move the Z-seam which would fix it.
The software cannot anticipate everything a user might do.  It is up to the user to adjust the settings to get a good result.
Any user can go through the settings and create a condition that doesn't work.  That doesn't mean there is a bug.  The model did slice and exactly reflected how you had it set up which resulted in one line being air printed.  Change one of several settings and the problem goes away.

mbenkmann, 2023-11-18:
Open your favorite word processor, type something, and go to File/Print... Then try to find a combination of settings that will cause that print to fail. You can't do it. Because paper printing is a mature technology. That's where 3D printing wants to be. I have found a combination of perfectly reasonably settings that cause the print to fail despite the fact that Cura could easily slice the model slightly differently. I am filing this report, together with a simplified model and a minimal set of settings I worked out to demonstrate the issue so that Cura can be improved and get closer to where it needs to be. I am confident that a small check in the Cura code can detect this situation and produce better slicing results.

Kathode-Ray, 2023-11-22:
> Open your favorite word processor, type something, and go to File/Print... Then try to find a combination of settings that will cause that print to fail. You can't do it. Because paper printing is a [...]

That's never going to happen. The reason is simple:

Anyone can get a print to fail by disabling support with a model that needs support to print succesfully.

mbenkmann, 2023-11-22:
This model DOES NOT require support to print.