Ultimaker/Cura #17356: Slicing Error

KillerTomatoe01, 2023-11-17:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

Ender 3

### Describe your model

Loyalist Emperors Children Space Marine

### Add your .zip here ⬇️

[CE3E3V2_Emperors children soldier.zip](https://github.com/Ultimaker/Cura/files/13395284/CE3E3V2_Emperors.children.soldier.zip)

GregValiant, 2023-11-17:
Thanks for the report.
That model has a lot of errors in it.  When surfaces flipped inside-out the slicer can't determine where the inside stops and the outside begins.
This is a view in Mesh Mixer.  The striped areas indicate problem areas.
I'll go ahead and close this.  If it won't slice after it's repaired then let me know here and I'll re-open it.  I hope you understand.
[image]

closed 2023-11-17 by GregValiant

KillerTomatoe01, 2023-11-17:
Ah ok cool, thanks for letting me know!