Ultimaker/Cura #17358: Using mesh modifier to control fuzzy skin only pushes fuzzies back

Smooth-Tii, 2023-11-17:
### Cura Version

5.3.1

### Operating System

Windows

### Printer

XTLW Climber 8 IDEX

### Reproduction steps

Used a mesh modifier to disable fuzzy skin so I could make a smooth logo to fyzzy skin print.
Another case was to use mesh modifier to disble fuzzy skin on inside parts, but results were the same. 

### Actual results

Instead of disabling the fuzzy skin the modifier only moved it back and the fuzzies were inside the print. 

### Expected results

Modifier should have disabled the fuzzy - not move it back. 

### Add your .zip and screenshots here ⬇️

Here are some shots of the behaviour. 
[image]
[image]

VipulDinesh, 2024-02-28:
Similar issues with fuzzy skin and mesh modifier (per model settings) while trying to test out the differences between different densities of fuzzy skin.

### **Cura Version**
5.6.0

### **Operating System**
Windows 11

### **Printer**
Elegoo Neptune 3 Plus

### **Reproduction steps**
1. Enable Fuzzy Skin for overall print.
2. Use a mesh modifier to print fuzzy skin with a different density for a second model.

### **Actual Results**
Original Fuzzy Skin gets pushed inside where mesh modifier is used and the modified fuzzy skin shows up for outer walls. This basically results in twice the number of walls being printed.

### **Expected Results**
Modifier should have overridden/replaced global fuzzy search, leaving the print with the same number of walls.

### **Screenshots**
Left Cube - Global Fuzzy Skin - 3 walls (Print Settings on the right)
Right Cube - Modifier Fuzzy Skin - 6 walls (Modified Settings on the left)

[image]

GregValiant, 2026-10-01:
Some work was done on fuzzy skin.  I don't know if it applies to this exact situation.
I'll close this as being stale.  If it is still a problem in 5.13.0 please make out a new bug report.  You can reference this report and you won't have to recreate the images.

closed 2026-10-01 by GregValiant