Ultimaker/Cura #17361: trouble slicing part

NoahsDad0904, 2023-11-18:
### Cura Version

5.4.0

### Operating System

Windows 11 Home

### Printer

AnyCubic

### Describe your model

Brake Rotor

### Add your .zip here ⬇️

[working-brake-assembly20220217-61-139jvfa.zip](https://github.com/Ultimaker/Cura/files/13398618/working-brake-assembly20220217-61-139jvfa.zip)

Brake Rotor doesn't want to slice

GregValiant, 2023-11-19:
Thanks for the report.
I didn't have any trouble slicing this is 5.5.0.  There is a bug in 5.4.0 that was worked on for 5.5.0 and it's a bit better.
There are three workarounds that can often fool the bug either separately or in a combination.

- Move the model on the Cura build plate.
- Rotate the model in Cura.
- Change the Initial Layer Height by a little bit.

This is a well known bug in 5.4.0 so I'll mark this as a duplicate.  The Cura team has a lot of examples of this to help chase down the bug so I'll go ahead and close this report.  I hope you understand.
[image]

closed 2023-11-19 by GregValiant

NoahsDad0904, 2023-11-19:
Thank you for helping me..Sent from my iPad