Ultimaker/Cura #17362: {JG}Knitted Stocking

Ted5855, 2023-11-18:
### Cura Version

5.4

### Operating System

windows 10

### Printer

JG Maker Artist D

### Describe your model

no

### Add your .zip here ⬇️

[{JG}_Knitted_Stocking_Pt1.zip](https://github.com/Ultimaker/Cura/files/13400240/JG._Knitted_Stocking_Pt1.zip)

GregValiant, 2023-11-19:
Yes.  It appears to be a stocking.  They often come in pairs.

Ted5855, 2023-11-25:
yes, but it could not slice. and I had other things that would not slice on the update

GregValiant, 2023-11-25:
You could have mentioned that when you filed the bug report.  We aren't mind readers.
There are some workarounds that alone or in combination can often fool the bug.

- Move the model to a different place on the Cura build plate.
- Rotate the model in Cura
- Change the Initial Layer Height by just a little bit (like .03).
- Toggle Generate Support on and off.

This bug has carried over into 5.5 and now 5.6beta.  It is being worked on by the Cura team.  I'll mark this as a duplicate and go ahead and close it as the team has lots of examples of this behavior.  I hope you understand.

closed 2023-11-25 by GregValiant