Ultimaker/Cura #17365: Cura 5.5.0 seems to lose it's place (XY) when printing multiple items.

rpweaverjr, 2023-11-18:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

Creality Ender 3 Pro with upgraded board (SGen V 1.0 , not SGen L) Marlin

### Reproduction steps

1. I created a clamp with Fusion 360. It consists of two sides, a magnet side and a metal bar side but they are basically the same design except for the space where the magnet or metal bar is attached.
2. I printed these one at a time with Cure 5.4.0 and they worked fine.
3. Installed 5.5.0
4. Sliced 2 each, 4 total, on the build plate using 5.5.0, estimated 12 hour print. (gcode attached)
5. Printed this overnight
6. The prints were skewed, especially after about 1/4" tall. The same offset is seen on all items like Cure skipped/lost or gained XY steps
7. The printed items are useless.
8. Preview does not show the shift.
[2Clamps.zip](https://github.com/Ultimaker/Cura/files/13401114/2Clamps.zip)

### Actual results

This file was printed one side at a time on the build plate from 5.4.0. Elephants foot is the only major issue.
[image]
This file was printed with 4 pieces on the build plate using 5.5.0
[image]
This is a single side print sliced with 5.5.0
[image]

### Expected results

The multiple prints should look like the single prints only replicated.
[TwoClampsClamp.zip](https://github.com/Ultimaker/Cura/files/13401349/TwoClampsClamp.zip)

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-11-19:
Thanks for the report and the files.
When you slice something in Cura and show the preview it is the data from the slicing engine.
But when you open a gcode file in Cura, a separate program reads the gcode into Cura to display it.  The preview of the gcode file when opened in Cura is a true and accurate representation of the toolpaths in the file.
You see here that the walls are nice and straight.  It looks like you had Adaptive Layers turned on.
[image]

As a double check, here is the same gcode file read into AutoCad.  Other than having created this gcode file, Cura is not involved in this.  The walls are straight.
Here is an iso view in Acad.
[image]

Here is an orthogonal front view.  The walls are all straight.
[image]

The problem you are experiencing would seem to be with the printer as the gcode is fine.  I have a high degree of confidence that the gcode file you posted would print fine on my own Ender 3 Pro.

I'll remove the bug label from this.  I'll leave it open for a while so you can respond here.

rpweaverjr, 2023-11-19:
Thank you for the reply. Initially, I thought it was the printer but how would the printer know there were multiple objects on the build plate. I am thinking that the travel between objects is too slow on the printer vs. the gcode. Cura set this at 250mm/sec. I am not sure how gcode is processed but my fear is that the travel is still in process when the new gcode command is attempted and skipped/lost because the previous command is not yet complete. This seems to only happen when I have multiple objects on the build plate. I just aborted a print, two objects, that was halfway through, 3.5 hours and plastic wasted, because the top seemed to be drifting away from the vertical. This would again result in useless objects. I changed the travel speed to 100mm/s and will try again.

rpweaverjr, 2023-11-19:
I didn't touch Adaptive Layers and don't know exactly what it does. I will turn it off if it is on and I can find it. (I found it and there is a check mark, is it defaulted? It looks like it is the default.)
I was thinking direct addressing vs relative addressing might be the problem but I don't even know if that is possible.

GregValiant, 2023-11-19:
The default is that "adaptive layers" is turned off.  If you happen to open a project file that has it turned on then that would turn it on, or it was turned on inadvertently.

_"... the travel is still in process when the new gcode command is attempted"_
That can never happen.  The printer/processor does one command at a time.  There is no "look ahead" function.

Cura knows how many "meshes" are on the build plate whether they are models or support blockers.  The speed going from one model to the next is at "travel speed".  If you have your travel speed at 250 that's how fast it would go.
For the layers above the "slow layers" your Gcode file shows a travel speed of F9000 which is 150mm/sec but your project file is 120mm/sec.  You made a change between making the gcode and creating the project file.  My normal travel speed is 150 and it has never been a problem.  On large models I go to 175.

I think you need to go over the printer mechanicals very carefully.

- Check all the trolley wheels.  You should be able to just turn them with two fingers.  If they are loose then things can wobble and make for inaccurate prints.  The print head should move on the X beam smoothly and not wobble.  The print bed is large and heavy and if it wobbles it can be a real problem.
- Make sure the hot end is tight and isn't wobbling.
- The belts on the X and Y should "twang like the strings on a bass guitar".
- The "X" beam should remain parallel during any move.  If you grab the right end and gently move it up and down does it feel firm?
- After about 1/2 hour of printer - pinch each stepper motor between two fingers.  Can you comfortably hold it?  If they are running too hot they can be glitchy and if they are running cold they may not make enough power.
- Did you "upgrade" the motherboard or did you order the printer with it?  If you did it you might want to take a quick look to make sure you have all the fans plugged into the proper ports.  The original Ender 3 Pro (like mine) had the motherboard fan wired with the layer cooling fan.  If the layer cooling fan wasn't running then the mainboard fan would never come on.  The stepper driver chips could get hot and get wonky.

You are using the basic "Dynamic" quality profile.  All the settings (except for Adaptive Layers) are the stock settings and should work fine.  For a test, try setting all the Acceleration settings to 500 (they are currently 1000).  That will slow down the starts and stops of the X and Y and may point to something specific.

When you print something like a ball there will be steps at each layer change.  "Adaptive Layers" works by changing the layer height by a bit so the steps are finer for features that approach horizontal and larger for things that are move vertical.  Because your models are basically identical, I'm sure it had no effect on what you are experiencing.  (When I saw the angled surfaces of your models I assumed that's why Adaptive was turned on because that's the sort of thing it's for.).

GregValiant, 2023-11-19:
I was just closing all those files and I noticed that your gcode shows as a 15hr print but my gcode created from your project shows a 7 1/2 hr print.  So something was radically different when you created that gcode but I'm unable to find it.

That's an oddity but I'm still confident this is a printer problem.

rpweaverjr, 2023-11-19:
I had to recreate the project since I didn't save it when I tried another test print setup. Sorry. BTW, changing the Travel seemed to help on the first attempt but the second was as bad as those pictured here. I am trying another test using multiple spiralized objects. I'm not sure that isn't an oxymoron. Turning off adaptive layers didn't help either.
Is there a way to home XY periodically, say every x number of layers? I was thinking that might reset the offset instead of letting it add up.

rpweaverjr, 2023-11-19:
""... the travel is still in process when the new gcode command is attempted"
That can never happen. The printer/processor does one command at a time. There is no "look ahead" function."

How does the processor know when the hardware has completed the last command? Time? There is no sensor to say it is complete. I am a retired career computer professional. I spent the last 6 years of my career in a manufacturing environment controlling manufacturing lines with PLCs to control the hardware. I always got feedback from the PLC when an operation was complete. I'm still not convinced that the XY doesn't get offset from the assumed position. Maybe it is the printer but how?
The belts seem to be tight. I'll recheck everything once the spiralized print completes. I did do a spiralized print with one object and the vertical surfaces were nearly perfect. I still don't understand why it makes a difference when printing one object vs printing multiple objects because the differences seem to be HUGE!

rpweaverjr, 2023-11-19:
The spiralized objects do not exhibit the XY errors although there are some anomalies, I assume, associated with the spiralize mode. There are lines of plastic joining the objects not included in the design. Because of that. I will join the two objects to eliminate the jumps between the objects that I suspect are the cause of the XY errors.
[image]
The front object was printed individually. Except for the plastic that is not in the design, the objects look virtually identical without the XY errors seen in multiple object prints.

GregValiant, 2023-11-19:
As you've seen - you can't print multiple objects in spiralize.  The print head has to leave one object to move to the next and it keeps extruding along the way.  For the same reason you can't spiralize any object that will develop "islands" during the print.

When the processor sees a command like G1 X100 Y100 it will figure out the number of steps to send to each motor that's involved.  It will send each motor the correct number of steps and at the right time so both motors leave their current position and arrive at the destination simultaneously.  That's true for any G0, G1, G2, G3 command.
An "all motors move" like G1 X100 Y100 Z10 E1.111 would be executed the same way.  It's "where am I now and where do I need to go next" and then everything gets there at once.

There really isn't any feedback across the USB.  I've written myself an app that monitors the printer while printing from the SD card.  There is some chatter across the USB but it won't include location information unless there is a G92 E0 in the gcode.  When G92 is encountered the printer will respond with the location (my printer was at the home offset position when I sent the G92):
X:0.00 Y:-5.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
ok
I can send M114 during a print and get that response with the current location..  By the time I get the response, the print has progressed and it's old news.

I'm an old toolie myself.  Here is a link to the Cliff's notes version of my app.  [Greg's SD Print Tool](https://drive.google.com/uc?export=download&id=1DY3xzdvzYuvAk8oSqvnwJnhjYSo-xUWG) It's Windows only and you will need a regular data USB cable to connect to the printer.  You will likely need to go into the Cura MarketPlace and disable "USB Printing" because it always grabs the USB port when Cura starts.
It's an un-signed app so if you decide to install it you may have to convince your anti-virus to allow it.
There aren't any instructions, but it ain't rocket science.  You can list the files on the SD card and then pick the one you want to print.  As the print is running all printer responses will show up in the text box.
In the image below the text box is showing the response to M503.
[image]

rpweaverjr, 2023-11-19:
The joined objects print completed with much better results then the stand alone objects.  There seems to be a bleed through of the bottom of the cavities. I've noticed this for a very long time that cavities bled through to the outside.  I'll bet it is the same cause.

Sent from my Verizon, Samsung Galaxy smartphone

[Print Tool]<https://user-images.githubusercontent.com/64202104/284080709-1e1bf4f2-346c-4837-a897-93d62261eac6.png>

rpweaverjr, 2023-11-19:
The inserts fit nicely into the cavities I made for them which was not the case for the multiple disconnected object prints. For this project, I am happy with these results but that still doesn't explain why the printer (Marlin) and Cura have different opinions of the XY position when printing disconnected multiple objects. From the spiralized prints, I really don't think it is a hardware problem. The outside surfaces are as good as I think can be expected by a bed slinger 3D printer.
Print results for connected objects. The exterior walls are much better with the objects connected although not perfect.
[image]
Inserts added. the screws are next. You can clearly see the joining wall I used. I have 2 more sets to print and I am done. I have 8 total magnetic clamps.
[image]

GregValiant, 2023-11-19:
_"There seems to be a bleed through of the bottom of the cavities"_
I don't know what that means.  Can you  post a picture of that?

It looks like you have your bed temperature at 60.  I wonder if it's really 60 or if it's running hot.  If you're printing PLA you can actually print it with a cold bed.
Try running at 60 for the Initial Layer and then let it drop to 40 for the rest of the print and see how it goes.  You could also try checking the bed temperature with a thermometer.  My thinking is that you are getting some heavy elephants foot and maybe the bed temperature isn't accurate.

GregValiant, 2026-09-10:
I'll go ahead and close this.

closed 2026-09-10 by GregValiant