Ultimaker/Cura #17367: Slicing failed CR-10 V2

arbarnum, 2023-11-19:
### Cura Version

5.2.1

### Operating System

Windows 11

### Printer

CR-10 V2 with upgrades

### Describe model location

center plate

### Describe your model

articulated dragon model (complex but no supports needed)

### Add your .zip here ⬇️

[CCR10_FlexiCrystalDragon.zip](https://github.com/Ultimaker/Cura/files/13402020/CCR10_FlexiCrystalDragon.zip)

Kathode-Ray, 2023-11-22:
Wow, that's a nice model! But it's very detailed, and I think that maybe part of the problem.

There's a limit to the smallest blob a 0.4mm nozzle can produce. It can't print very tiny details. Have a look at some of the makes on Cults3D for example. Most of the small crystals are gone and end up as blobs. And, the tips of the pointed crystals are infinitely small, and that's impossible to print. But I can get it to slice succesfully by disabling `Print Thin Walls`.

### 0.4mm nozzle example

[image]

I would recommend to use a 0.2mm nozzle for this model, to get a better end result. The next example is also with `Print Thin Walls` disabled. Even then, it takes ages to slice.

### 0.2mm nozzle example

[image]

Normally, Cura ignores the too fine details, because they can't be printed anyway. If you have `Print Thin Walls` set to enabled, this tells Cura to have a look at those details and to print anything larger than your `Minimum Feature Size` of 0.105mm. That's just a fraction of a millimeter and since there are so many details in this model, it probably causes the slice to fail.

Hope this helps!

Best regards,

Raymond

MariMakes, 2024-01-18:
Hi @arbarnum,

Quick update from our side 👋
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪

Even with the changes we made in the latest versions of Cura we still see that your model crashes. 🫤 But we have added your project file to our analyzing tool and found there were at least 10 crashes that were similar to yours, which helps us to narrow down what the cause is of your crash so we can resolve this for you and others.
Hopefully, we can find a solution soon 🤞

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/17438

closed 2024-01-18 by MariMakes

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0