Ultimaker/Cura #17370: Cant´t slice bigger Things

gelberlemmy, 2023-11-19:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Anycubic Kobra 2

### Name abnormal settings

default profil is ok

### Describe model location

nor rotate

### Describe your model

no

### Add your .zip here ⬇️

The Model and my Profil. When i slice little Models, everything is okay.

[PLA 0.2 Sonder.zip](https://github.com/Ultimaker/Cura/files/13402631/PLA.0.2.Sonder.zip)

GregValiant, 2023-11-19:
Thanks for the report.
I didn't have any trouble with it.  It might have something to do with your settings.  Use the "File | Save Project" command to create a full project file.  Zip that file and post the zip folder here.  I get the "Profile is not compatible with AnyCubic Kobra" warning when I try to import that profile you posted.
There is a bug that started in 5.4.0 and was worked on (but isn't fixed in 5.5.0) that can keep models from slicing.  You can try one or a combination of:

- Move the model around in Cura.  Get it away from the Cura midpoint.
- Rotate the model in Cura.  90° CCW has worked for me.
- Change the Initial Layer Height by like 0.03.

This was with the base PLA profile of a Kobra.
[image]

gelberlemmy, 2023-11-19:
Okay thanks for your quick response. Here are my saved Projekt of this model. I have testet your ideas, but nothing works for me. 

[CFFFP_main-7.2.zip](https://github.com/Ultimaker/Cura/files/13403371/CFFFP_main-7.2.zip)

GregValiant, 2023-11-19:
I can't get it to slice with the "PLA Sonder" profile but it will slice with the Draft profile.  
Switch to that profile and then touch up your temperature settings.  It seems like you should be able to slice and print it.

closed 2025-03-10 by gelberlemmy