Ultimaker/Cura #17373: mid poly conan helmet

dokuoireru3, 2023-11-19:
### Cura Version

5.4.0

### Operating System

windows 10 pro

### Printer

elegoo neptune 3 max

### Name abnormal settings

no

### Describe model location

no

### Describe your model

non of the above

### Add your .zip here ⬇️

here is the link via wetransfer. i have reduced the the poly count from high to mid to see if it would help. but the slicer is still giving me the same error. the file size is now smaller then other files i have that my slicer works fine with. not sure what the issue is. thank you for your help.

<html>
<body>

https://wetransfer.com/downloads/081e8f67188026b2832d4e420892236a20231119164841/6907b2af64d2ce4546071d613ebddbbc20231119164841/80f653
--
1 item

</body>
</html>[https://wetransfer.com/downloads/081e8f67188026b2832d4e420892236a20231119164841/6907b2af64d2ce4546071d613ebddbbc20231119164841/80f653](https://wetransfer.com/downloads/081e8f67188026b2832d4e420892236a20231119164841/6907b2af64d2ce4546071d613ebddbbc20231119164841/80f653?trk=TRN_TDL_01&utm_campaign=TRN_TDL_01&utm_medium=email&utm_source=sendgrid)
1 item

GregValiant, 2023-11-20:
Thanks for the report.
If you have Mesh Tools loaded (from the MarketPlace) you would get this warning.
[image]

The red surfaces facing upwards indicate surfaces that are inside out.  The model needs repair 
This particular model is made up of 42 different pieces.  There is an excellent chance that it wasn't assembled correctly and that leaves gaps in the mesh that are making it "not watertight".
This is in Mesh Mixer.  The two pieces with red stripes show a lot of errors and will not slice, or will slice incorrectly.
[image]

A surprising number of helmets show up here in bug reports.  I haven't seen this one before and it is an extremely ambitious model to print.  With your settings I'm guessing it will be a 7 or 8 day print.  Right now I've had the repaired model slicing for about 1 1/2 hours.
And then I went to bed.
When I woke up I found that this will take just over 6 days to print.  Quicker than I thought.
[image]

So is this a bug or is it the model?  It's hard to tell.  I'll leave the bug label on it but if it is a bug it's the same old bug that has been reported numerous times so I will mark this as a duplicate and go ahead and close it.  I hope you understand.

closed 2023-11-20 by GregValiant

dokuoireru3, 2023-11-21:
so you think you could send me back the repair model? or let me know how you repaired it? pretty please !!! thank you

GregValiant, 2023-11-21:
MS 3d Builder is available in the Microsoft store.  It isn't intuitive, but it's pretty good.  Open the model and then click on the Insert button.  The repair routine runs automatically.