Ultimaker/Cura #17374: Failed

r3b3ll8, 2023-11-19:
### Cura Version

5.5.0

### Operating System

MacOS BigSur

### Printer

DreamerNX

### Name abnormal settings

no

### Describe model location

no

### Describe your model

yes

### Add your .zip here ⬇️

[DNX_GFK Masthalter v1.3mf.zip](https://github.com/Ultimaker/Cura/files/13404316/DNX_GFK.Masthalter.v1.3mf.zip)

GregValiant, 2023-11-20:
Thanks for the report.
The project sliced for me.  Try moving the model on the Cura build plate, or rotating the model or maybe changing the Initial Layer Height by a little bit.
I'll mark this as a duplicate and the Cura team will take a look.
This with no changes to your settings.
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic