Ultimaker/Cura #17375: Can't Open .stl Files or .gcode Files From Folder

SkeleBones, 2023-11-19:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Anycubic Kobra

### Reproduction steps

1. Set Ultimaker Cura 5.5.0 as default app to open .stl & .gcode files.
1. Double-click .stl & .gcode file or right-click then click open. 

### Actual results

Does not open any .stl & .gcode file. 

### Expected results

It should open Ultimaker Cura with the model open on it when double-clicking the file or right-clicking and clicking open after setting UltiMaker Cura as the default app to open said file.

### Add your .zip and screenshots here ⬇️

To open a .stl or .gcode file, I must first open Ultimaker Cura then drag the file onto the 3D printer plate.  On the previous stable version (5.4.0), this worked as described.

GregValiant, 2023-11-19:
Thanks for the report.
Double-clicking on a file invokes Windows to either start the correct app, or to send the file to the correct app.  It would seem that something isn't telling Windows that STL and Gcode files should default to Cura.
I have several versions of Cura loaded and double-clicking a 3mf will always start 4.13.1 because that was the last time I explained things to Windows.  You need to specify which Cura version you want to use.

Try re-setting the Windows default by right-clicking on an STL and selecting properties.  Select the "Change" button next to "Opens with" and then keep scrolling down and clicking on Show More Apps until it lets you browse the hard drive.  Go to your Cura installation folder and for a 5.x version select "UltiMaker-Cura.exe".  (On 4.x versions it was Cura.exe.)

This is actually a Windows problem so I'll remove the bug label.  I hope you understand.  I'll leave this open for a bit so you can respond here.

SkeleBones, 2023-11-20:
I have selected the correct version being 5.5.0 but it doesn't allow them to open.

GregValiant, 2023-11-20:
This might be just a Windows 11 problem as it should work.  It certainly works on my Win 10 computer.
Maybe someone on the Cura team is aware of this happening with Windows 11 so I'll put the bug label back on this so they notice.

closed 2025-03-14 by SkeleBones