Ultimaker/Cura #17379: 'Alternate Wall Directions' doesn't alternate wall directions while 'Surface Mode' is set to Surface

ElBartoUno, 2023-11-20:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Custom Printer with ReprapFirmware

### Reproduction steps

1. Import model to print area
2. Set 'Surface Mode' to Surface
3. Check 'Alternate Wall Directions'
4. Slice and preview sliced model
5. While sliding through layer proces it's visable that all walls are printed in the same direction

### Actual results

1. The setting that is suppose to alternate the wall direction does not alternate the wall direction between layers 

### Expected results

1. While a model is printed with surface mode, the wall direction of the current should be the opposite of the previous layer

### Add your .zip and screenshots here ⬇️

[CilinderPrint_BothModes.zip](https://github.com/Ultimaker/Cura/files/13417157/CilinderPrint_BothModes.zip)

GregValiant, 2023-11-21:
Thanks for the report.
I can duplicate this behavior.
"Surface" mode is closely linked to the behavior of "Spiralize" which is always in the same direction.

It's in the Experimental section so maybe the setting itself needs to be tweaked.  It's also possible that it's incompatible with Surface mode and should not be available is "Surface" is selected (and vice versa).

The Cura team will take a look.

ElBartoUno, 2023-11-21:
Hi @MariMakes ,

For the company I work at, we primarily print with Surface Mode. We print with a clay like substance with cellulose as reinforcing fiber, that really tends to move while it's printed (as seen in [this video](https://youtu.be/D0bz-JHZJPM?si=nOj5lhyrkiaASpab&t=20)).
Surface Mode allows us to import .STL files of surfaces and the print path will follow the surface as modeled. As hopefully is made clear with the following pictures:
[image]
[image]

The above printing method is simply not possible without Surface Mode set to Surface, as far as I know.
The Alternate Wall Directions setting will hopefully allow us to print objects where tensions don't build up that much in corners.

I hope I made it all clear and it's simple enough to implement it. 

Thanks! :)

ElBartoUno, 2023-11-21:
Just to clarify: It's a self-intersecting surface model that allows the printhead to move through already printed lines. That's possible because the material remains soft until it's dried.

GregValiant, 2026-10-01:
I'll close this as stale.

closed 2026-10-01 by GregValiant