Ultimaker/Cura #17380: Gridfinity Base 5x5

wimprest, 2023-11-20:
### Cura Version

5.4.0

### Operating System

Win 11

### Printer

Ender 3 Pro

### Name abnormal settings

There are ABL settings in the pre-print script, but it shouldn't affect the slicing?  I've done a dozen or so on this current Cura build

### Describe model location

Centered

### Describe your model

5x5 Gridfinity base.

### Add your .zip here ⬇️

[grid_5x5.zip](https://github.com/Ultimaker/Cura/files/13418085/grid_5x5.zip)

GregValiant, 2023-11-21:
Thanks for the report.
There are more and more bug reports coming in that involve Win11.
Here is your model with my Ender 3 Pro active but on Win10.
[image]

You can try rotating the model 90° or changing the Initial Layer Height by a little bit.  Sometimes those can fool the bug.
I'll mark this as a duplicate and the Cura team will take a look.

wimprest, 2023-11-21:
Thanks for the reply! If I run into this in the future, I'll try those
steps first. I like having a little bit of troubleshooting ability before
bringing in the big guns like you!

GregValiant, 2023-11-21:
You can say that if it worked.
I don't know about this Win11 thing.  The OP system may just be getting more popular, or Cura may not be quite set up to deal with it yet.  There does seem to be some glitchyness there though.

wimprest, 2023-12-27:
Hey Greg,

Update:  the slicing error is due to the large model pushing the
auto-generated skirt to push into the no-go area of the printer bed.

When I set adhesion to "none" the slicing worked just fine.  When I added
it back in? Slicing error.

Perhaps adding a printable adhesion check into the slicing protocol will
help you generate more helpful error messages?
Either way, thanks for working with me on this.

Thanks,
Wim

GregValiant, 2023-12-27:
A big model like that doesn't need a lot of skirt.  1 loop at 2mm from the model (Skirt Distance) would be enough to get the filament flowing.
Another thing that can affect the printable area is "Avoid printed parts when traveling" because the nozzle moves outside the print and if it goes into the no-print zone then the model won't slice.
A change was made between 5.4 and 5.5 (I think) and now if any little bit of skirt will fit on the bed then the model will slice.  If none of the skirt will make it onto the bed then it still won't slice.
[image]

GregValiant, 2026-10-01:
I'll close this as stale.

closed 2026-10-01 by GregValiant