Ultimaker/Cura #17381: f86 fuselage7

baeshen23, 2023-11-20:
### Cura Version

5.5.0

### Operating System

windows 11

### Printer

creality ender 3 s1 pro

### Name abnormal settings

y

### Describe model location

y

### Describe your model

other parts sliced perfect

### Add your .zip here ⬇️

[CE3S1PRO_F-86A fuselage 7.zip](https://github.com/Ultimaker/Cura/files/13418122/CE3S1PRO_F-86A.fuselage.7.zip)

GregValiant, 2023-11-20:
Thanks for the report.
Within the project file you posted the "Surface Mode" is set to "normal".  This type of model is made up of surfaces that have zero thickness and so Surface Mode usually needs to be set to "Surface".  I made that change and it sliced in both 5.5 and 5.6beta.  I think I see some issues though.  Notice the vertical tube here.
[image]

It is 100% disconnected from the model (as are all the other instances of those tubes) and a lot of the side ribbing is disconnected from the main outer skin.  I think an attempt to print this the way I set it up would fail and it would collapse into a pile of separate parts.
[image]

This model is a group of dis-connected surfaces.  It will slice in "Surface" so I don't know if that could be considered a bug.  Not slicing correctly would seem to have more to do with the design.  I'm sure you ran into this in the wings and other pieces.  How was the ribbing connected to the outer skins?  The instructions for the fuselage slice were followed exactly?

All my RC models were balsa and monokote so I took a brute force approach to power.  If it was too heavy for a 35 put a 60 on it.  Then I'd plant it in the ground and bring the pieces home in a bucket.
I'll leave the bug label on this and the Cura team will take a look.  Good luck with it (even though I'm more of an F-16 guy).

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic