Ultimaker/Cura #17382: title

FilipViaene2, 2023-11-20:
### Cura Version

5.6.0 beta1

### Operating System

windows10

### Printer

creality cr-x pro dual

### Describe model location

center

### Describe your model

round model,

### Add your .zip here ⬇️

[post_1.zip](https://github.com/Ultimaker/Cura/files/13418356/post_1.zip)

GregValiant, 2023-11-20:
Thanks for the report.
Go to the Cura Marketplace and load the Mesh Tools plugin.  When you load a model Mesh Tools will check it for errors.
When if finds errors you will be notified and in some cases the model will be painted in funky polka-dots like your models on the right.  The models on the left were repaired in MS 3D Builder.
[image]

Model errors make it difficult for the slicer to tell where the outside of a model ends and the inside begins.
Here are the repaired models sliced.
[image]

I'll remove the bug label from this as it appears to be a model issue.  I'll leave this open for a bit so you can respond here.

GregValiant, 2026-09-10:
I'll close this.

closed 2026-09-10 by GregValiant