Ultimaker/Cura #17383: keeps crashing

AmmaPysz, 2023-11-20:
### Cura Version

5.5.0

### Operating System

windows 10

### Printer

ender 3 neo

### Reproduction steps

uploaded bluey stl from printables two different blueys different developers and cura keeps not responding 

### Actual results

keeps not responding then closing the app

### Expected results

cura to go not responding and crash

### Add your .zip and screenshots here ⬇️

[CE3PRO Bluey Model Sitting STL Export Generic PLA lw 0.45mm lh 0.2mm if 20% ext1 210C bed 60C.zip](https://github.com/Ultimaker/Cura/files/13418661/CE3PRO.Bluey.Model.Sitting.STL.Export.Generic.PLA.lw.0.45mm.lh.0.2mm.if.20.ext1.210C.bed.60C.zip)
[cura.log.zip](https://github.com/Ultimaker/Cura/files/13418668/cura.log.zip)

GregValiant, 2023-11-21:
Thanks for the report.
I can't get it to slice either.  I'll mark this as a duplicate report and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic