Ultimaker/Cura #17385: Slicing Failed, lithophane, .2 mm nozzle, extra fine profile

staltux, 2023-11-21:
### Cura Version

5.5.0

### Operating System

Slackware Linux 15

### Printer

Elegoo Neptune 3

### Name abnormal settings

Extra fine with .2 nozzle, the fine with .2 works fine

### Describe model location

slice with 180 degrees

### Describe your model

is a lithophane, slice with the profile "fine"

### Add your .zip here ⬇️

[project.zip](https://github.com/Ultimaker/Cura/files/13420811/project.zip)

GregValiant, 2023-11-21:
HAH.  Try to fool me will ya...
With the:

- Support Z distance at 0.39
- Support Top Distance at 0.39
- Support Bottom Distance at 0.40

With the bed adhesion set to "None" this can be duplicated.
With normal Top and Bottom Distances the problem doesn't occur.  With any Bed Adhesion other than "None" this doesn't occur.

So @staltux if you change those silly Top and Bottom Support Distance settings to something reasonable the problem goes away.  They should be exact multiples of the Layer Height.

If you set the Top Distance to .399 and the bottom distance to .39 then with Bed Adhesion set to None the initial layer height will be 0.001.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic