Ultimaker/Cura #17389: couche initial trop rapide

loroff, 2023-11-21:
### Cura Version

5.5.0

### Operating System

windows 10

### Printer

lk4 pro

### Reproduction steps

vitesse d impression couche initial demandé a 20 mm/s

### Actual results

des mur de la couche initiale ne s imprime pas a la vitesse demandé (trop rapide )

### Expected results

-

### Add your .zip and screenshots here ⬇️

[LLK4PRO_NOEL.stl0.zip](https://github.com/Ultimaker/Cura/files/13424289/LLK4PRO_NOEL.stl0.zip)

loroff, 2023-11-22:
[image]

GregValiant, 2023-11-23:
Thanks for the report.
I know I can duplicate this one.  This was bad in 5.5 and it's terrible in 5.6beta2.  The problem seems to be the "Flow Equalization Ratio" in the speed settings.  Setting that to "0" will fix the problem.

I'll mark this as a duplicate but I'll leave it open for now. (also see #17324 and #17010 ).

loroff, 2023-11-23:
[image]

loroff, 2023-11-23:
meme avec le rapport d egalisation de debit a 0% j ai une difference de vitesse sur la couche initial

GregValiant, 2026-10-01:
I haven't seen this problem in the last couple of versions so I'll close this.

closed 2026-10-01 by GregValiant