Ultimaker/Cura #17390: [5.6.0-beta.1] Support Z-distance feedback: Flow isn't reduced to compensate for thinner support layers leading to overextrusion

printingotb, 2023-11-21:
### Cura Version

5.6.0-beta.1

### Operating System

win10

### Printer

Custom fff

### Reproduction steps

See project
Please also see my other reports submitted at the same time, some are related.

### Actual results

It appears as if the initial support layer and the top most support layer are both over extruded, as if being printed with the flow of an ordinary support layer. This can be seen visually on the top, from the side of the structure and on the underside where lines become much wider. As a result, structure 4 comes out much thicker than structure 5 when measured with calipers. The extra filament has forced itself around the nozzle and even lifted it so some degree. Though it in theory should only be a difference of 0.01mm its actually almost 0.2mm thicker. Reading the G-code for those structures, the Z-coordinates looks correct at Z1.61 and Z1.6.
I'm using a support interface density of 97%. Having a sparser support structure would be more forgiving, as is the case at layer 0.

Over-extruded Support structures: Layer height=0.2mm.
Support Z-distance:
1. 0.20mm smooth
2. 0.30mm slightly over-extruded
3. 0.35mm moderately over-extruded
4. 0.39mm massively over-extruded
5. 0.40mm smooth

### Expected results

Flow reduced to match the reduction in layer height of the affected support lines.

### Add your .zip and screenshots here ⬇️

(the circles seen on 2-4 are explained in a separate report)
[image]
From 1-5. "over-extrusion" increases the closer it gets to next full layer distance. At 0.4mm the issue disappears again as it is a full layer multiple with no modified thickness.
[image]

From Gcode topmost support layer:

1
Z-distance: 0.2mm
;LAYER:8
M204 S1500
;TYPE:SUPPORT-INTERFACE
G1 F600 Z1.8

2
0.30mm
;LAYER:8
M204 S1500
;TYPE:SUPPORT-INTERFACE
G1 F600 Z1.7

3
0.35mm
;LAYER:8
M204 S1500
;TYPE:SUPPORT-INTERFACE
G1 F600 Z1.65

4
0.39mm
;LAYER:8
M204 S1500
;TYPE:SUPPORT-INTERFACE
G1 F600 Z1.61

5
0.40mm
;LAYER:7
M204 S1500
;TYPE:SUPPORT-INTERFACE
G1 F600 Z1.60

Vandresc, 2023-11-27:
Hi Jonas, could you please share a project with us? Thank you

Vandresc, 2023-12-01:
Deva, see ticket on the backlog for our internal reference CURA-11424.

GregValiant, 2026-10-01:
I'll close this as being stale.  I it is still a problem in 5.13.0 this can be reopened.

closed 2026-10-01 by GregValiant