Ultimaker/Cura #17392: [5.6.0-beta.1] Support Z-distance feedback: Normal support and Interface support is being placed incorrectly

printingotb, 2023-11-21:
### Cura Version

5.6.0-beta.1

### Operating System

win10

### Printer

Custom fff

### Reproduction steps

See project.

Please also see my other reports submitted at the same time, some are related.

### Actual results

Normal support is being placed around and on top of support interface. This also happens with support floor turned off and support brim turned off. Bed adhesion set to none.

### Expected results

Normal support and interface to be placed correctly

### Add your .zip and screenshots here ⬇️

[Normal vs interface glitch.3mf.zip](https://github.com/Ultimaker/Cura/files/13424500/Normal.vs.interface.glitch.3mf.zip)
[image]
[image]
(Reducing Support horizontal expansion removes the extra normal support lines on layer 0 but not the ones resting on top of the interface. I had set it high to make test-prints easier to separate)

Vandresc, 2023-11-28:
Hmm, I cannot seem to reproduce the support on the interface when the Z distance is 0.21. We will have a release soon, could you please re-check this in the stable? Thank you

printingotb, 2023-11-30:
Yeah it still happens in 5.6.0. (you have to change the Z-distance in the project from 0.2 to anything between 0.21 and 0.39)

Vandresc, 2023-12-01:
I made a ticket for the backlog, Devs see CURA-11423 for our internal reference.

casperlamboo, 2024-03-19:
issue should be solved here https://github.com/Ultimaker/CuraEngine/pull/2049, the fix will be in our next release, 5.7, scheduled to be released in a few weeks.

casperlamboo, 2024-03-19:
solves https://github.com/Ultimaker/Cura/issues/17392

closed 2024-03-19 by casperlamboo