Ultimaker/Cura #17393: [5.6.0-beta.1] Support Z-distance feedback: Support thickness of 0.01mm on layer 0

printingotb, 2023-11-21:
### Cura Version

5.6.0-beta.1

### Operating System

win10

### Printer

Custom fff

### Reproduction steps

see project. The model is a 10x10x0.2mm square raised 1mm above the buildplate. It has 3 layers of support.
Layer height and initial layer height is 0.2mm. Support Z-distance is 0.39mm

Please also see my other reports submitted at the same time, some are related.

### Actual results

The Z coordinate at layer 0 is 0.01. Wrong?
Also, the normal support lines on layer 0 gets reduced like I would expect the topmost support interface layer to be. I assume because it "believes" it's the last layer of support? 
Note, I discovered that the extraneous normal support lines disappear when I reduce my support horizontal expansion. I had set it extra high to make testing easier.

### Expected results

Z-coordinate on layer 0 should be same as initial layer height?
I believe printing a 0.01mm thick first layer risk damaging the buildplate, especially if the bed is leveled a bit too close to the nozzle. (This is why I moved the support structures upwards in my script-attempt instead of downwards https://github.com/Ultimaker/Cura/issues/6105#issuecomment-1613301213).

### Add your .zip and screenshots here ⬇️

[Layer 0 Z.3mf.zip](https://github.com/Ultimaker/Cura/files/13424625/Layer.0.Z.3mf.zip)
[image]
[image]
[image]

GregValiant, 2023-11-21:
Thanks for the report.
When I slice you file I get this at Layer:0
;--------------------------------------
;
G92 E0
G92 E0
G1 F1500 E-0.8
;LAYER_COUNT:6
;LAYER:0
M107
M204 S1333
M205 X5 Y5
**G1 F600 Z0.4** >>>>> This is Initial Layer Height of 9.2 + ZHop Height of 0.2
G0 F6000 X103.834 Y103.834 **Z0.4**
M204 S1000
;TYPE:SUPPORT
G1 F600 **Z0.2**  >>>>>This is "Hop down"
G1 F1500 E0
G1 F1373 X103.948 Y103.803 E0.00425
G1 F1488.8 X116 Y103.8 E0.40407
and at the beginning of the file is **;MINZ:0.2**

I don't understand that move to Z=.01 at all.  I do notice that your initial Z move is to 0.21.  Do you have some sort of Z Offset adjustment going on that isn't in the project file?  It is notable that the 0.21 is Initial Layer Height + the 0.01 that you show in the gcode.

So far I cannot duplicate the problem using your project file.

GregValiant, 2026-10-01:
The current version is 5.13.0  and this problem has been fixed.

closed 2026-10-01 by GregValiant