Ultimaker/Cura #17404: Multiple Bridge Layer settings are being ignored

brickmaster32000, 2023-11-22:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Ultimaker S5

### Reproduction steps

1. Load bridge test stls
2. Enable bridge setting
3. Enable multiple layer bridge settings
4. Set flow rate of first three bridge layers to 25%
5. Slice

### Actual results

The first bridge layer's flow rate is set to 25%, the remaining bridge layers ignore their settings and get set to 100%

### Expected results

The flow rate for the first three bridge layers should be set to 25%

### Add your .zip and screenshots here ⬇️

[UMS5_bridge20mm.zip](https://github.com/Ultimaker/Cura/files/13443949/UMS5_bridge20mm.zip)
[image]
[image]

GregValiant, 2023-11-22:
Thanks for the report.
Looks good to me.  Exactly how it is set up in Cura.
For your homework you have to figure out why that's true.  (Hint:  Look carefully at the names of the Cura settings and then look at the preview again.)

I'll go ahead and remove the bug label and close this as this is expected behavior.  I hope you understand.

closed 2023-11-22 by GregValiant

brickmaster32000, 2023-11-24:
The whole point of the feature is to make adjustments the layers above a bridge to make them print properly. That means both the skin and the walls need to be adjusted.  I get that you can cover your ass by saying that as it currently is it only refers to the skin but that just means that the intended behavior is wrong. Was this feature intended to not perform its function?