Ultimaker/Cura #17407: Incorrectly adding M105 M109 when 'print one at a time' selected, but only between first and second model.

marsmanp, 2023-11-23:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

Anycubic Mega Pro

### Reproduction steps

Slice with minimum of 3 models on bed and 'one at a time' setting turned on

### Actual results

The gcode includes a lines with M105 and M109 S200 between the first and second object, but does not between the second and third objects.  
[image]

### Expected results

There should not be any M105 or M109 lines between the objects

### Add your .zip and screenshots here ⬇️

[pawprint 3 small.zip](https://github.com/Ultimaker/Cura/files/13445514/pawprint.3.small.zip)

GregValiant, 2023-11-24:
Thanks for the report.
I can duplicate this.
The M105 is "Report Temperature" which will have no effect.
The M109 won't hurt anything but often causes a delay of about 10 seconds before the print resumes.
Those lines are supposed to be added when the "Initial Layer Print Temperature" is different than the "Print Temperature" but in your case that was not true and so they should not have been added.

The Cura team will take a look.

GregValiant, 2025-01-15:
I just checked in Cura 5.9.0 and this appears to be fixed.  I'll close this.

closed 2025-01-15 by GregValiant