Ultimaker/Cura #17410: Train_model_slice_fail

acg0501, 2023-11-23:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

Ultimaker 3

### Name abnormal settings

The settings are default

### Describe model location

Even rotated the model fails to slice.

### Describe your model

The model has intricate details un top of a plate and needs support

### Add your .zip here ⬇️

[UM3_155 Kabinenkesselwand.zip](https://github.com/Ultimaker/Cura/files/13451400/UM3_155.Kabinenkesselwand.zip)

5axes, 2023-11-26:
As many issues, the origine  : **Print Thin walls**

GregValiant, 2023-11-26:
@acg0501 Thanks for the report.
I agree with @5axes.  Turning off "Print Thin walls" allows it to slice.

The Cura team will take a look.

MariMakes, 2024-01-18:
Hi @acg0501,

Quick update from our side 👋
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪

Even with the changes we made in the latest versions of Cura we still see that your model crashes. 🫤 But we have added your project file to our analyzing tool and found there were at least 10 crashes that were similar to yours, which helps us to narrow down what the cause is of your crash so we can resolve this for you and others.
Hopefully, we can find a solution soon 🤞

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/17438

closed 2024-01-18 by MariMakes

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0