Ultimaker/Cura #17412: STL failing to slice with unexpected error.

T3chDad, 2023-11-23:
### Cura Version

5.4

### Operating System

Windows 10 22h2

### Printer

FLSUN Super Racer

### Describe model location

Model will not slice regardless of orientation.

### Describe your model

Never had a success slicing this STL.

### Add your .zip here ⬇️

[FSR_SwordOfFathoms(Prepped1.1).zip](https://github.com/Ultimaker/Cura/files/13452827/FSR_SwordOfFathoms.Prepped1.1.zip)

Vandresc, 2024-01-19:
This specific model still fails for me with the latest fixes from ThomasRham and w/o having support enabled, so not quite sure it is related to tree support.

HellAholic, 2024-02-16:
Small update. Issue has been resolved and the fix will be included in the next release.
[image]

closed 2024-02-16 by HellAholic

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0