Ultimaker/Cura #17417: K1 Max profile please!

dandontknow1, 2023-11-24:
### Is your feature request related to a problem?

There's no Cura profile for Creality's latest and greatest?! i mean, what's going on with that?!

### Describe the solution you'd like

A cura profile for the K1 max which can utilise all its features.

### Describe alternatives you've considered

Creality Print. This made me sad.

### Affected users and/or printers

I should imagine many people would benefit from this!

Changowen, 2023-11-28:
see this: https://www.youtube.com/watch?v=Vua0OfaRg2g

dandontknow1, 2023-11-28:
I think most of us would like a proper native profile in Cura like with other printers.

itay-grudev, 2024-01-02:
Could one of you please take a look at #17871 and verify independently that the provided printer definition works as expected.

Also please note that Cura won't enable the side-fan by default.

dandontknow1, 2024-01-02:
Nice one. If I knew how to try test code, I would but I'm not too clever with GitHub and coding. Sorry.

cadbob, 2024-03-21:
i used to love using cura but since i got the k1max , i have had to use orca, orca is good but was so used to cura and would like to use again.

PR #17871 (2024-05-03): K1 Max Printer Definition
# Description

Adds printer definition for Creality K1 Max.

Please note that, when printing at High Speed, Cura won't enable the side fan which will result in poor cooling and deformation. Make sure to manually enable it before printing.

## Type of change

- [x] Printer definition file(s)

# How Has This Been Tested?

- [x] Tested at very high speed - 600mm/s.
- [x] Tested at high speed - 300mm/s.

**Test Configuration**:
* Operating System: Windows / Linux

# Checklist:
- [x] My code follows the style guidelines of this project as described in [UltiMaker Meta](http [... cut]
closed 2024-05-03 by saumyaj3