Ultimaker/Cura #17418: Vase mode doesn't work properly

00raq00, 2023-11-24:
### Cura Version

5.6 beta 2

### Operating System

Windows 10

### Printer

Artillery Genius/Hornet

### Reproduction steps

1. open model
https://www.printables.com/model/334709-cute-christmas-tree-light/files

2. slice using outer contour option 

### Actual results

on bottom part there is missing hole for tealight 
[image]

### Expected results

exist hole for tealight 

[image]

### Add your .zip and screenshots here ⬇️

[AH_tree (1)2.zip](https://github.com/Ultimaker/Cura/files/13459243/AH_tree.1.2.zip)

Asterchades, 2023-11-25:
Vase mode (or `Spiralize Outer Contour` to use its in-program name) automatically enables the `Remove All Holes` mesh fix, which is doing exactly what it's supposed to do. Hollow models cause problems for the generation routines, resulting in the path jumping back and forth through the middle and ultimately causing the print to fail, so this is by design. This model, for example, would be one of those that fails without the mesh fix enabled by virtue of the hollow underneath.

This is an unfortunate side-effect of the transition to the newer slicing engine of Cura 5. Cura 4 would indeed give the results you're expecting, and for this particular project it may be worth installing an older version (eg [4.13.1](https://github.com/Ultimaker/Cura/releases/tag/4.13.1)). Note that the two installations can co-exist on the same system without interfering with one another.

It is possible to modify this model to work under Cura 5. The indent on the bottom needs to be reduced in depth such that it matches the Top/Bottom Height you've specified (default 0.4mm). This would let you disable the Remove All Holes mesh fix, which would in turn maintain the hole in the bottom.

Note that neither of these work-arounds would fix the "eyes" on the side. These have 90° overhangs and flat surfaces which would have been designed with both infill and skin in mind, so may be formed very poorly in vase mode. It also won't correct the inverted geometry on the rest of the model, but that doesn't seem to be causing any problems.

I'll be removing the bug tag from this as it's (unfortunately) deliberate, but leaving it open for further discussion.

smartavionics, 2023-11-25:
You may wish to try my Cura variant (https://github.com/smartavionics/Cura/releases) which is based on Cura 4 and has some additional improvements to the vase mode (and elsewhere). It's a pity that they broke vase mode in Cura 5 but, hey, that's progress for you.

GregValiant, 2026-09-10:
There is a way to fool UM Cura into making a hole there using a cutting mesh "Bottom Thickness" tall.
I'll close this.

closed 2026-09-10 by GregValiant