Ultimaker/Cura #17427: yakupbalaban@gmail.com

yakupbalaban, 2023-11-25:
### Cura Version

5.5.0

### Operating System

windows 11

### Printer

jg maker artist d pro

### Name abnormal settings

custom

### Describe model location

any posisiton was not worked

### Describe your model

human heart produced by blender wireframe format.

### Add your .zip here ⬇️

[Slicing error 02 of holed heart duble extrusion.zip](https://github.com/Ultimaker/Cura/files/13464135/Slicing.error.02.of.holed.heart.duble.extrusion.zip)

steljwagh, 2023-12-14:
I suggest you change the title of this issue.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

yakupbalaban, 2024-04-03:
Thank you for your answer. I will try the 5.7 version on my irregular
models, but I would like you to know that the 4 versions of cura did not
cause any errors in slicing anatomical models. For this reason, I am still
using the 4.6 version.

3 Nis 2024 Çar 15:38 tarihinde HellAholic ***@***.***> şunu
yazdı:

HellAholic, 2024-04-03:
Hey @yakupbalaban,
You can continue using the versions that you're most comfortable with. That is the nature of the open source software. This is a free tool available in different variations and as such you are free to choose the version that suits your needs best.