Ultimaker/Cura #17428: Cant open with more then one screen

Pitti999, 2023-11-25:
### Cura Version

5.5.0 -5.6 Beta 2

### Operating System

Windows 11

### Printer

Anycubic 4mac pro 2.0

### Reproduction steps

Opening the program with 2 screens turned on.

### Actual results

Just shows a white screen, and the program wont answer.

### Expected results

I ecpect it to turn on... :D

### Add your .zip and screenshots here ⬇️

[image]
[image]

Pitti999, 2023-11-26:
After a test with another slicers, I can conlude it has something to do with my setup.

fieldOfView, 2023-11-26:
The other silicer runs into the same problem that Cura is running into. It is (probably) an interplay between your GPU driver and the GUI framework the applications use.

Here's something you can try: Disable all but one of the monitors and then launch Cura. Go into the preferences and uncheck the "Restore window position on start" checkbox. Now Cura should start, always on your primary display, even when multiple monitors are enabled.

Burgerpower01, 2023-12-15:
Ran into the same issue, you could try the following (it worked for me at least...):

I had to edit the cura.cfg located at C:\Users\<username-here>\AppData\Roaming\cura\5.5 and delete the following lines:
```
window_left = 3530
window_top = 19
window_width = 1718
window_height = 1360 
```

Hope this helps

GregValiant, 2026-10-01:
I'll go ahead and close this although every once in a while a slightly different version of this bug shows up.

closed 2026-10-01 by GregValiant