Ultimaker/Cura #17432: Top layers effects bottom layers

Thisismydigitalself, 2023-11-25:
### Cura Version

5.5

### Operating System

Win10

### Printer

TLM

### Reproduction steps

These holes starts 1mm from bottom.
[image]
This is how first layers looks like when top layers are less than 3.
[image]
This is how first layers looks like when top layers are set to 5
[image]

Why Top surface skin layers effects the bottom layers?

### Actual results

Why Top surface skin layers effects the bottom layers?

### Expected results

Top layers should not effect bottom layers.

### Add your .zip and screenshots here ⬇️

These holes starts 1mm from bottom.
[image]
This is how first layers looks like when top layers are less than 3.
[image]
This is how first layers looks like when top layers are set to 5
[image]

Why Top surface skin layers effects the bottom layers?

GregValiant, 2023-11-26:
Use the "File | Save Project" command, zip the 3mf file, and post the zip folder here.
It isn't possible to troubleshoot this without a project file.
It would appear that @5axes has seen this problem as well.  I haven't noticed it.

5axes, 2023-11-26:
Yes Top Skin Surface Layers takes precedence over Bottom Layer . I think we should have at least one Bottom Layer.
[image]
[Cube_Hole.zip](https://github.com/Ultimaker/Cura/files/13468184/Cube_Hole.zip)

tkocou, 2023-11-26:
I can confirm this problem. I changed the bottom surface layer from lines to concentric, but left the top surface as lines. After slicing the model and started the print, the bottom surface still printed with lines. I changed the top surface to be concentric as well. Sliced the model and the first layer printed concentric as intended.

GregValiant, 2026-09-29:
Time to close this.

closed 2026-09-29 by GregValiant