Ultimaker/Cura #17433: Calibration cube labels X/Y are swapped.

maXe72, 2023-11-26:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

Creality Ender S1 Pro / Clipper

### Reproduction steps

1. Menü Settings -> Part for Calibration -> Add a calibration Cube

### Actual results

The sliced result is a 20mm calibration cube, but the model is rotated by 90 degree. 
So the X and Y axis letters on the sliced and printed cube are swaped against each other.
If you measure the X-Axis labed side of the cube printed you are actually measuring the Y-Axis printed and vice versa.

### Expected results

The X labeled side should represent the X-Axis printed.
The Y labeled side should represent the Y-Axis printed.

### Add your .zip and screenshots here ⬇️

[image]

5axes, 2023-11-26:
We could also consider that the  X represent the length of the Axis we want to calibrate.
[image]

maXe72, 2023-11-26:
Yes, but it is counterintuitve and oposide way of measurement  in comparision to the z-axis

5axes, 2023-11-26:
Yes, but when you measure the Z-Axis You don't see the Z mark and you don't care as you know that it's the Z direction.  Anyway it's not a bug it's a feature and it will not change.

closed 2023-11-26 by maXe72