Ultimaker/Cura #17434: Missing Printer Weedo X40 from this release

chasrobertson, 2023-11-26:
### Cura Version

5.5

### Operating System

windows 11

### Printer

generic

### Reproduction steps

Unable to find the printer definition for the Weedo X40

### Actual results

This definition was on all the other releases of Cura

### Expected results

How do I copy the printer from the 5.4 version?

### Add your .zip and screenshots here ⬇️

None required

GregValiant, 2023-11-26:
My install of 5.5.0 shows all the Weedo X40 files are present, but the printer isn't available in the list in Cura.  I looked through the files and I don't see why it isn't in the list.  It isn't throwing an error when Cura tries to load it.

@MariMakes of the Cura team may take a look.

GregValiant, 2026-09-29:
Time to close this.

closed 2026-09-29 by GregValiant