Ultimaker/Cura #17435: Cura 5.5. Windoes x64 adds spurious extra single line to skirt shortly after print starts, this is reproducible. Probably on layer 0 change

paulbauwens, 2023-11-26:
### Cura Version

5.5.0

### Operating System

Windows 10 Pro 64-bit   GPU , i53470, NVIDIA Geforce GT1030

### Printer

Ender 3 Pro + SKR Mini E3 V1.2 board and Marlin 2.0.9.7 LTS FW

### Reproduction steps

No special steps, happens on any print.  I am using a 3-line skirt, more is usually not necessary
Printer is 100% stable

### Actual results

No harm done, but should not be the case.  

### Expected results

Just print a 3-line skirt

### Add your .zip and screenshots here ⬇️

Looks like a minor bug

paulbauwens, 2023-11-26:
The actual problem seems to have been eliminated, f

The problem is that Cura adds a spurious extra line to the skirt at the beginning of the print, looks like it happens on the 0 to 1 layer change or maybe 1 -2 change

GregValiant, 2026-09-29:
Time to close this.

closed 2026-09-29 by GregValiant