Ultimaker/Cura #17437: CCR10_Octopus Plush by BODY3D ( Fix )

Brumblebeemastr, 2023-11-26:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

CR-10

### Describe your model

plush Articulating Octopus

### Add your .zip here ⬇️

[Octopus Plush by BODY3D ( Fix ).zip](https://github.com/Ultimaker/Cura/files/13467561/Octopus.Plush.by.BODY3D.Fix.zip)
it has work before  and just all of a sudden stopped working.

GregValiant, 2023-11-26:
Thanks for the report.
When I turned off "Print Thin Walls" this sliced for me.  Be patient.  This took a long time to finish slicing.

I'll mark this as a duplicate and the Cura team will take a look.

[image]

Brumblebeemastr, 2023-11-27:
So the thing is. That is turned off and failing to slice

Respectfully, Michael Brummer

image.png (view on web)<https://github.com/Ultimaker/Cura/assets/64202104/69b12004-e0a9-4d38-957f-9dea24f189eb>

GregValiant, 2026-09-29:
Time to close this.

closed 2026-09-29 by GregValiant