Ultimaker/Cura #17438: Unable to slice ANYTHINg with "Print Thin Walls" Activated

Kiefels2, 2023-11-26:
### Cura Version

All Versions

### Operating System

Windows 10

### Printer

Kingroon KP3s 

### Name abnormal settings

No

### Describe model location

No

### Describe your model

Yes

### Add your .zip here ⬇️

I had this issue before and it went away but now it's back again.
I cannot slice any models that I have sliced succesfully before, with "Print Thin Walls" checked.
Rotating models and moving them off centre does not rectify this issue either.
I have tested the same file on all versions Beta included, since Vers 5.3 Christmas Vers Alpha up to the Beta 2 Vers 5.6.0
All the same.
Worrying thing is, everything sliced fine the previous day, with no issues.
No soft or hardware was added since the last successful slice.
[Air Line Nozzle.zip](https://github.com/Ultimaker/Cura/files/13467669/Air.Line.Nozzle.zip)

Until the issue came back, my PC had not been rebooted or updated as far as I am aware.

Backend exited abnormally with return code 3221225477! is the error that I get on all versions

GregValiant, 2023-11-26:
Thanks for the report.
I can duplicate this one.  That error code is familiar as well.
@rburema may be interested in this one.

The Cura team will take a look.  This is a duplicate of other reports regarding Print Thin Walls and I'll mark it as such.

rburema, 2023-11-27:
That's odd. The engine is unfortunately not completely deterministic (yet), but it shouldn't have this big of an alteration between runs (where it completely works one day and doesn't the next). Especially strange that this seems to happen across multiple versions. Could you attach (the rest of) the logs? You should be able to find the location of those via the Cura help-menu.

@GregValiant is there anything else in your logs?

MariMakes, 2024-01-18:
Hi @Kiefels2,

Quick update from our side 👋
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪

Even with the changes we made in the latest versions of Cura we still see that your model crashes. 🫤 But we have added your project file to our analyzing tool and found there were at least 10 crashes that were similar to yours, which helps us to narrow down what the cause is of your crash so we can resolve this for you and others.
Hopefully, we can find a solution soon 🤞

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0