Ultimaker/Cura #17439: No projects are slicing in Cura

JZ4wJVJastejz, 2023-11-26:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Monoprice Select Mini V2

### Describe model location

Center of build plate

### Describe your model

Video game Charatcer. Model was changed in scale to fit for build area. nothing will slice. No projects. Including projects already used and printed successfully. 

### Add your .zip here ⬇️

[Darksiders 2  Death - 3325777.zip](https://github.com/Ultimaker/Cura/files/13468202/Darksiders.2.Death.-.3325777.zip)

GregValiant, 2023-11-26:
Thanks for the report.
When I turned off "Print Thin Walls" this sliced for me.  It took a while to slice but it did finish.

[image]

MariMakes, 2024-01-18:
Hi @JZ4wJVJastejz,

Quick update from our side 👋
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪

Even with the changes we made in the latest versions of Cura we still see that your model crashes. 🫤 But we have added your project file to our analyzing tool and found there were at least 8 crashes that were similar to yours, which helps us to narrow down what the cause is of your crash so we can resolve this for you and others.
Hopefully, we can find a solution soon 🤞

I'll be closing this issue as a duplicate but you can follow the progress here https://github.com/Ultimaker/Cura/issues/16417

closed 2024-01-18 by MariMakes

HellAholic, 2024-02-16:
Small update,
Issue is resolved in CURA-11570 and will be part of the next Cura release.