Ultimaker/Cura #17455: Cura 5.4.0 Slice failed

richardcdean, 2023-11-27:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Prusa i3 Mk3.Mk3s

### Name abnormal settings

No

### Describe model location

No.  I used Slic3r STL fix but use Cura to slice.

### Describe your model

Yes, Yes, Yes.   From Thingavers, printed ok many times.

### Add your .zip here ⬇️

Available through my email at contractcam@hotmail.com. Richard C. Dean

I cannot tell if the attachments actually attached.
[Bicycle-snap-sheet-1.0slic3r.zip](https://github.com/Ultimaker/Cura/files/13481002/Bicycle-snap-sheet-1.0slic3r.zip)
[Bicycle-snap-sheet-1.0slic3r.zip](https://github.com/Ultimaker/Cura/files/13481004/Bicycle-snap-sheet-1.0slic3r.zip)

richardcdean, 2023-11-27:
[Bicycle-snap-sheet-1.0slic3r.zip](https://github.com/Ultimaker/Cura/files/13481018/Bicycle-snap-sheet-1.0slic3r.zip)

richardcdean, 2023-11-27:
Cura 5.5.0 also fails

richardcdean, 2023-11-27:
Caused by scaling up X from 91mm width to 150mm.

5axes, 2023-12-07:
Stl model is useless to analyse an issue , You should post the project file (3mf file). No issue for me.

[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

richardcdean, 2024-04-06:
I cannot recreate the problem using 5.7.0.

Thanks for you effort,

Richard C. Dean BsEE, MsMT.