Ultimaker/Cura #17456: Slicing fail on dungeon wall

dokcal, 2023-11-28:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Modified Ender-3 Pro

### Name abnormal settings

Added Hole expansion

### Describe your model

I have sliced and printed this model, but it was rotated 90 degrees, and didn't have hole expansion.

### Add your .zip here ⬇️

[DangerousDungeons_LED_Brazier_25mm_x3FAIL.zip](https://github.com/Ultimaker/Cura/files/13482108/DangerousDungeons_LED_Brazier_25mm_x3FAIL.zip)

5axes, 2023-12-07:
An other **Print Thin Wall** issue ? That seems logical for this kind of part.
[image]

MariMakes, 2024-01-18:
Hi @dokcal 

Quick update from our side 👋
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪

Even with the changes we made in the latest versions of Cura we still see that your model crashes. 🫤 But we have added your project file to our analyzing tool so we can start collecting similar crashes so we can resolve this for you and others.
Hopefully, we can find a solution soon 🤞

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic